mendelcollege-robotics/robotica #26: nieuwe wielen + barrel jack oplader

Jorns57, 2025-03-25:
wielen met frank enzo

robosoccer, 2025-03-27:
ook een teensy en twee seeed xiao

robosoccer, 2025-03-27:
https://gtfrobots.com/product/50mm-diameter-4mm-hole-size-omni-wheel/           6x

robosoccer, 2025-03-27:
https://www.tinytronics.nl/en/development-boards/microcontroller-boards/others/seeed-studio-seeeduino-xiao-cortex-m0-samd21 2x                                             https://www.tinytronics.nl/en/development-boards/microcontroller-boards/teensy/teensy-4.1-without-ethernet

Jorns57, 2025-03-30:
Barreljack oplader -> Jorn neemt mee vanuit huis

Jorns57, 2025-04-03:
Do 11-04 : wielen uit elkaar halen -> kwaliteit bepalen

robosoccer, 2025-04-04:
teensy, seeed's en openMV kabel zijn binnen gekomen

Jorns57, 2025-04-06:
ik heb thuis de juiste set screws (m4*~5mm) voor de wielen, die hoeven we dus niet te bestellen

closed 2025-04-10 by Jorns57